ProstateBRP/BRPRobot2021 #2: Problem building the SNR files

FaridTavakol, 2021-11-29:
The CMakeLists.txt includes a folder called testing which is not included in the repo. This would create an error while building the project. Please either modify the CMakeLists.txt or add the required files to the directory.

rebeccalisk, 2021-11-29:
Sorry, not sure why the Testing folder isn't showing up when I push my changes to GitHub... my local copy has a testing folder within SNR/client, but I will investigate why it isn't uploading.

rebeccalisk, 2021-11-29:
So for some reason, when I name the folder "Testing," it refuses to be pushed to the GitHub. Renaming it to BridgeTesting fixed the issue. Let me know if you can build the code now

FaridTavakol, 2021-11-29:
@rebeccalisk Thanks it solved the issue! The problem with testing file not being added to the remote repo was that the Testing was in the .gitignore file.

closed 2021-11-29 by FaridTavakol