ProstateBRP/BRPRobot2021 #3: Slicer module does not open a server

FaridTavakol, 2021-11-29:
Hi @lisaepfl,  I was trying to test the simulator and the Slicer module seems to fail to create a port for the client to be connected to. Could you investigate this issue please?

lisaepfl, 2021-11-29:
Hi @FaridTavakol! Sure I will look into it but I cannot build because of the following error: BRPRobot2021/WPI/server/RobotCommunicationBase.h:73:7: error: function definition does not declare parameters
  int connect{0}; // Shows the status of socket connection
      ^
/Users/lisa/Desktop/BRPRobot2021/WPI/server/main.cxx:192:22: error: no member named 'connect' in 'RobotSimulatorPhaseBase'
    (*currentPhase)->connect = 1;
    ~~~~~~~~~~~~~~~  ^
/Users/lisa/Desktop/BRPRobot2021/WPI/server/main.cxx:196:29: error: no member named 'connect' in 'RobotSimulatorPhaseBase'
    while ((*currentPhase)->connect)
           ~~~~~~~~~~~~~~~  ^
3 errors generated.

Could you please let me know if you have addressed this issue?

FaridTavakol, 2021-11-29:
I don't get this error while building. @rebeccalisk could you test and confirm this issue?

rebeccalisk, 2021-11-29:
I actually get a different error while trying to build the WPI folder -- 

/home/rebeccalisk/BRPRobot2021/WPI/server/RobotSimulatorStartUpPhase.cxx: In constructor ‘RobotSimulatorStartUpPhase::RobotSimulatorStartUpPhase()’:
/home/rebeccalisk/BRPRobot2021/WPI/server/RobotSimulatorStartUpPhase.cxx:33:9: error: ‘class RobotSimulatorStartUpPhase’ has no member named ‘RegisterDefectType’
   33 |   this->RegisterDefectType("DNR", "Device-not-ready in START_UP phase.");
      |         ^~~~~~~~~~~~~~~~~~
/home/rebeccalisk/BRPRobot2021/WPI/server/RobotSimulatorStartUpPhase.cxx: In member function ‘virtual int RobotSimulatorStartUpPhase::Initialize()’:
/home/rebeccalisk/BRPRobot2021/WPI/server/RobotSimulatorStartUpPhase.cxx:47:13: error: ‘class RobotSimulatorStartUpPhase’ has no member named ‘GetDefectStatus’
   47 |   if (this->GetDefectStatus("DNR"))
      |             ^~~~~~~~~~~~~~~
make[2]: *** [WPI/server/CMakeFiles/RobotTestSimulator.dir/build.make:102: WPI/server/CMakeFiles/RobotTestSimulator.dir/RobotSimulatorStartUpPhase.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:171: WPI/server/CMakeFiles/RobotTestSimulator.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

FaridTavakol, 2021-11-29:
@rebeccalisk Could you pull the latest commit?

rebeccalisk, 2021-11-29:
Sorry, it works now. Will check if the Slicer module creates a client port on my end

rebeccalisk, 2021-11-29:
Port creation works for me. @FaridTavakol are you sure you are testing with a port # that is closed? If 18944 isn't working and you haven't tried to switch to a new port, maybe try that?

FaridTavakol, 2021-11-29:
`AttributeError: module 'slicer' has no attribute 'vtkMRMLIGTLConnectorNode'
That's the type of error that I get. Maybe there are some issues with my Slicer/OIGTL installation. Thanks for checking. We could run a test tomorrow if you are free.
`

rebeccalisk, 2021-11-29:
I am free all afternoon or before 11 am to run a test. What would work for you two?
The only thing I can think of is that your SlicerOpenIGTLink installation has some problems as you mentioned, since I don't think that should be a code issue especially if Lisa's and mine works. Maybe you could try reinstalling the OIGTLink extension if you haven't already?

FaridTavakol, 2021-11-29:
Yeah, I believe the issue is on my side. Let's shoot for a 11 AM tomorrow to test the system before the meeting. if it works for Lisa as well.

closed 2021-11-29 by FaridTavakol

rebeccalisk, 2021-11-29:
Real quick- Lisa and I have a lab meeting at 11, can we do 10?

FaridTavakol, 2021-11-29:
Sure I can do 10! See you then!