ProstateBRP/BRPRobot2021 #4: Send time in the START_UP phase

FaridTavakol, 2021-12-01:
Add time stamp in the START_UP message sent from the Navigation software to the Robot. This will update the robot clock with the real world clock and helps maintain log in the robot of the time each important step has been initiated.

rebeccalisk, 2021-12-09:
Hey Farid -- sorry, just now seeing this! 
Since this isn't in the protocol, can you clarify what format the timestamp should be in? 
For the message "STRING(CMD_XXXX, START_UP)," should the timestamp be appended to the CMD_XXXX string? Or something else?

FaridTavakol, 2021-12-09:
That is a good idea! We can actually incorporate time in each message MSG_XXX send from the Navigation to the robot.
This can include CLB_XXX and CMD_XXX TGT_XXX and etc. 
In this case we both can keep a log of the time when each of these commands were sent and save it in a log.txt file for any future debugging and records. What do you think about that?

closed 2022-01-31 by rebeccalisk