ProstateBRP/BRPRobot2021 #8: Issue with the initial connectivity

FaridTavakol, 2022-01-17:
The initial connection between the robot and the navigation software does not connect and requires disconnection of the navigation software and reconnecting it. 
The issue seems to go away upon also when the slicer module disconnects and another port is used.

FaridTavakol, 2022-01-17:
Upon further investigation, one issue might be that the Slicer module does not recognize the change from Start UP phase to Start Up phase. This results in an error shown in the Slicer terminal when the Application is running. This requires the user to close Slicer and reopen it in order to make the issue go away.

rebeccalisk, 2022-01-19:
Hey Farid -- What do you mean by the Start UP vs. Start Up phases? My understanding is that there should be only one phase, START_UP. I haven't gotten any errors in the Slicer terminal, would you be able to send me a screenshot of what you see?

FaridTavakol, 2022-01-19:
> 
[image]

FaridTavakol, 2022-01-19:
In the above screen shot, the Slicer module was first opened on port 18944. Then due to not being able to send the Start-up message to the robot, the port was closed and another port 18945 was opened. This time upon clicking on the Start-up, the error shown in the bottom of the above image was seen in the python console. This is probably the fact that Slicer was keeping the state in its memory and upon multiple clicking on the same button it throws an error which should be addressed in this instance.

rebeccalisk, 2022-01-19:
Ohhh. Thank you! Something is up with the timestamp in the acknowledgement it seems, which is preventing the command and acknowledgement from matching up. I’m at a doctors appt right now but I should be able to fix that easily soon!

rebeccalisk, 2022-01-19:
I’m not exactly sure why it’s different upon successive start ups, but I’ll investigate. Thanks so much

closed 2022-01-31 by rebeccalisk