ProstateBRP/BRPRobot2021 #22: Request and Update robot position upon entering the Planning Workphase

FaridTavakol, 2023-04-10:
The robot position should be updated in the slicer upon entering the planning workphase.
This is because in the previous calibration state, the calibration matrix is sent to the robot, and that causes the robot's position to potentially shift wrt the imager. Therefore, in the planning, it would be beneficial to request the current position of the robot upon entering.

franklinwk, 2023-05-02:
Would this be covered by the continuous query for Current Position that is toggled on separately from the phases? Is there another robot position update you're referring to?

FaridTavakol, 2023-05-02:
> Would this be covered by the continuous query for Current Position that is toggled on separately from the phases? Is there another robot position update you're referring to?

Yes, it would but only if the query is turned on. Note that internally the robot already has updated the current pose of the robot when the calibration is sent. This issue addresses the lack of visual update (re-render the current position of the robot by performing a single position query without the need to toggle the current pose button) on the Slicer side when the calibration is sent to the robot. Moreover, we could query the robot's position each time a new verified calibration is sent to the robot.